me-no-dev/arduino-esp32fs-plugin #1: SPIFFS.begin failed to mount after uploading sketch data folder

ErinQvnm, 2017-10-05:
It successfully upload sketch data folder, here is the console output:

`[SPIFFS] data   : D:\Dropbox\Trial\TestSPIFFS\TestSPIFFS\data
[SPIFFS] start  : 2691072
[SPIFFS] size   : 1468
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/test.txt

[SPIFFS] upload : C:\Users\INFINI~1\AppData\Local\Temp\arduino_build_382024/TestSPIFFS.spiffs.bin
[SPIFFS] address: 2691072
[SPIFFS] port   : COM4
[SPIFFS] speed  : 921600
[SPIFFS] mode   : dio
[SPIFFS] freq   : 80m

esptool.py v2.1-beta1

Connecting........_

Chip is ESP32D0WDQ6 (revision 0)

Uploading stub...

Running stub...

Stub running...

Changing baud rate to 921600

Changed.

Configuring flash size...

Auto-detected Flash size: 4MB

Compressed 1503232 bytes to 2109...

Writing at 0x00291000... (100 %)
Wrote 1503232 bytes (2109 compressed) at 0x00291000 in 0.0 seconds (effective 353700.0 kbit/s)...

Hash of data verified.

Leaving...

Hard resetting...`

Then i tried to open the file with this simple code :

`#include <FS.h>
#include <SPIFFS.h>

void setup() {
	Serial.begin(921600);
	if (SPIFFS.begin()) {
		Serial.println("SPIFFS MOUNTED");
	}
	else {
		Serial.println("SPIFFS MOUNT FAIL");
	}
	if (SPIFFS.exists("/test.txt")) {
		Serial.println("FILE EXIST");
	}
	else {
		Serial.println("FILE NOT EXIST");
	}
}

void loop() {
  
}`

Can't mount SPIFFS, error : 

`E (372) SPIFFS: mount failed, -10025
[E][SPIFFS.cpp:47] begin(): Mounting SPIFFS failed! Error: -1
SPIFFS MOUNT FAIL
[E][vfs_api.cpp:22] open(): File system is not mounted
FILE NOT EXIST`

I can mount SPIFFS again using `SPIFFS.begin(true)` but ofc all files is deleted. Please help fix. Thanks

me-no-dev, 2017-10-05:
Mkspiffs is broken not the plugin. Surely will fix after the holiday ;) mkspiffs is not something that i can release. Thanks

me-no-dev, 2017-10-08:
mkspiffs fix is in this PR: https://github.com/igrr/mkspiffs/pull/31

ErinQvnm, 2017-10-11:
Ok, thanks.

me-no-dev, 2017-10-13:
all done :) run get.py/get.exe in arduino/tools to get the new version of mkspiffs

closed 2017-10-13 by me-no-dev

tferrin, 2017-10-20:
In case others trip over this same problem, I could not get this fix to work with the new mkspiffs installed in the Arduino/tools directory on my OSX system.  Instead I had to run "get.py mkspiffs" from the Arduino/hardware/espressif/esp32/tools directory.  After that I re-ran the Data Upload tool and it successfully uploaded my data files and SPIFFS.begin() not longer gets a error.

lefedor, 2018-01-05:
Had an error
E (2669) SPIFFS: mount failed, -10025
solved by this topic