me-no-dev/arduino-esp32fs-plugin #2: Documentation issues (e.g., "mkspiffs not found", broken links)

MartyMacGyver, 2017-10-08:
Arduino 1.8.4 on Windows with latest Arduino-ESP32.

In attempting this I get the error "SPIFFS Error: mkspiffs not found!" I see another issue here where you mention you can't distribute `mkspiffs` (as well as notation in the commit history on Arduino-ESP32).

Is the proper `mkspiffs` available elsewhere for the user to install separately? It'd be useful info to have in the README as to where that tool is and how to install it.

Also, the README has a broken link to the release page as well as a broken example snapshot link.

me-no-dev, 2017-10-08:
Yeah... I made the plugin, tested it out and found mkspiffs not working. So I kinda did not mention it's existence anywhere yet (except to test in gitter) until we fix that mkspiffs issue. mkspiffs will come with Arduino through get.exe/get.py just like the toolchain and all.

MartyMacGyver, 2017-10-08:
Thanks for the info - I presume it will be some variant of the one at https://github.com/igrr/mkspiffs/releases (which would be useful for trying out in the interim).

me-no-dev, 2017-10-08:
mkspiffs fix is in this PR: https://github.com/igrr/mkspiffs/pull/31

MartyMacGyver, 2017-10-08:
I understand now - thanks for the clarification! (And for the plugins!)

me-no-dev, 2017-10-13:
@MartyMacGyver the new mkspiffs is released :) update through get.py/get.exe

closed 2017-10-13 by me-no-dev

MartyMacGyver, 2017-10-13:
Thanks for the update!