me-no-dev/arduino-esp32fs-plugin #3: SPIFFS_write error(-10010): unknown

vyruz1986, 2017-10-16:
Thanks so much for releasing this tool, I was waiting on this to continue work on my project ([FlyballETS](https://github.com/vyruz1986/FlyballETS-Software))

I have made an angular v5 web app using Angular-CLI which needs to go on the SPIFFS filesystem. The angular-cli build generates the necessary files for production use. I have uploaded a data.zip file containing the output of  this production build: [Link](https://1drv.ms/f/s!AvhWSkEmBKJHlO4Hjjb6m6H-5sdyWA)

When I put the contents of this zip file in a /data/ folder of my arduino sketch and run the ESP32 Sketch Data Upload, I get the following error:

```
[SPIFFS] data   : C:\Users\alexg\OneDrive\Projects\FlyballETS-Software\data
[SPIFFS] start  : 2691072
[SPIFFS] size   : 1468
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/3rdpartylicenses.txt
/assets/mdi.svg
/favicon.ico
/glyphicons-halflings-regular.448c34a56d699c29117a.woff2

SPIFFS_write error(-10010): unknown
error adding file!
SPIFFS Create Failed!
```

I did some testing and it seems the file name length is the problem, I tried just uploading one file from the package and then started renaming it using trial and error:
```
NOK	: inline.a44fd682ff3096930d6e.bundle.js
OK	: inline.a44fd6823693.bundle.js
OK	: inline.a44fd68236933.bundle.js
NOK	: inline.a44fd682369333.bundle.js
```
The 3rd line is 30 characters, 4th line is 31 chars.

me-no-dev, 2017-10-17:
Yup :) SPIFFS file names can be 32 bytes max (including starting slash and zero at the end). This is a known limitation :)

closed 2017-10-17 by me-no-dev

vyruz1986, 2017-10-17:
Thanks for clarifying.
Better return error from the tool would be a nice improvement though 🙂