me-no-dev/arduino-esp32fs-plugin #4: Creating SPIFFS crashe

holema, 2017-10-17:
I installed the SPIFFS Toll like described in the installation part.

Arduino: 1.6.8
actual Arduino-ESP32 Core.
I installed the Tool in the \Documents\Arduino\hardware\espressif\esp32\tools folder

After starting the toll from the Arduino IDE I get this error:

Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.Sketch.getBuildPath()Ljava/io/File;
	at com.esp32.mkspiffs.ESP32FS.getBuildFolderPath(ESP32FS.java:118)
	at com.esp32.mkspiffs.ESP32FS.createAndUpload(ESP32FS.java:319)
	at com.esp32.mkspiffs.ESP32FS.run(ESP32FS.java:378)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

me-no-dev, 2017-10-17:
try with newer IDE :)

holema, 2017-10-17:
Thanks @me-no-dev with the Arduino version 1.8.5 it is working.