me-no-dev/arduino-esp32fs-plugin #5: Arduino IDE: SPIFFS Not Supported on arduino-esp32

kristsm, 2018-01-23:
Hi! Came to problem where Arduino IDE does not allow to upload to esp32 . When pushing "ESP32 Sketch Data Upload" it just warns about:  SPIFFS Not Supported on arduino-esp32.  Tried Arduino 1.8.5. and 1.85  (I am on Windows 10) with no luck. Compilation and upload of firmware works fine.

bospre, 2018-01-28:
I had the same error until I upgraded to the latest version
https://github.com/espressif/arduino-esp32

gggatelier, 2018-05-13:
Please change the folder name where esp32 is placed.
Arduino / hardware / espressif / arduino-esp32 
to
Arduino / hardware / espressif / esp32

I got the following error.
SPIFFS Not Supported on ESP 32
I changed the folder name from ESP32 to esp32 and it worked normally.

arnold-b, 2019-12-21:
Hey. I Also have this problem