me-no-dev/arduino-esp32fs-plugin #6: SPIFFS + ESP32 Sketch Data Upload Problem

smacyas, 2018-01-23:
Hardware:
Board:	?ESP32 Dev Module?
Core Installation/update date:	latest
IDE name:	Arduino IDE
Flash Frequency:	40Mhz?
Upload Speed:	115200

Description:
Hello.
I need help.
I have WROOM-32 ESP32 module.
In folder of my ino-project I created folder ‘data’ and put into it some files: css, html, txt by ESP32 Sketch Data Upload (https://github.com/me-no-dev/arduino-esp32fs-plugin), I have uploaded them successfully to my module.
When I try to read file from SPIFFS (previously mounted SPIFFS, of course) (latest arduino-esp32, https://github.com/espressif/arduino-esp32), listDir(SPIFFS, "/", 0) - OK; but while I use readFile(SPIFFS, "/style.css"); there is some trash, but no correct data. When I use write and read file, which was NOT uploaded by ESP32 Sketch Data Upload, it is OK.

By the way, I noticed string: flash read err, 1000 ets_main.c 371

Any ideas?
Thanks.

Sketch:
Debug Messages:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun 8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58

Starting device...
ESP32 Chip ID: 9238022
FreeHeap: 126324
ChipRevision: 1
CpuFreqMHz: 240
SdkVersion: v3.1-dev-239-g1c3dd23f-dirty
FlashChipSize: 4194304
FlashChipSpeed: 40000000
FlashChipMode: 2

SPIFFS Mount succesfull:
Listing directory: /
FILE: /door.png SIZE: 846
FILE: /favicon.ico SIZE: 239
FILE: /index.html SIZE: 4875
FILE: /logo.png SIZE: 5075
FILE: /profile.png SIZE: 898
FILE: /style.css SIZE: 5844
FILE: /style.txt SIZE: 101
FILE: /variables.txt SIZE: 41

Started WiFi Station Terminal Access (STA) regim
Current SSID: myssid

ESP32 MAC STA: [D][WiFiGeneric.cpp:293] _eventCallback(): Event: 2 - STA_START
30:AE:A4:06:F6:8C
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 1 - SCAN_DONE
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 7 - STA_GOT_IP

WiFi connected
WiFi channel: 1
WiFi RSSI: -56
192.168.200.162
Reading file: /style.css
Read from file: яяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяя

bospre, 2018-01-28:
Hello,
unfortunatly I have to confirm this behaviour.
If I just upload a file, I get garbage when reading the file.
If I upload the SPIFFS_Test.ino and it runs once with the line
 testFileIO(SPIFFS, "/test.txt");
not commented. the file isn't any longer corrupted.
Any idea what goes wrong here?

marhul, 2018-01-28:
Hi,
I had the same behaviour. And then I found this:
https://www.mgo-tec.com/blog-entry-esp32-spiffs-easy-websocket-library.html
When I change Arduino-esp32 to commit from 19/12/2017 everything is working.

smacyas, 2018-01-29:
Yes, I also thought that in December 2017 it worked, but after upgrading at the end of December it stopped. I hope this bug will be fixed.

hpwit, 2018-02-02:
@marhul Could you please tell me how to get the commit from 19/12 
thank you
rgds
yves

rfestag, 2018-02-02:
@hpwit:
Try this:
```
cd ~/Arduino/hardware/espressif/esp32
#I determined the checkout by running `git log`, and finding the last commit on 12/19/1027
git checkout 70d0d4648766cea047613062ed50a9ecfc9de31f
```
It seems to be behaving better now for me as well. Not sure if that means this library needs to be updated, or there was a bug introduced in arduino-esp32.

vindolin, 2018-02-02:
It's something in the static library tools/sdk/lib/libspiffs.a.
I replaced it with the one from 70d0d4648766cea047613062ed50a9ecfc9de31f and everything worked again.

baggior, 2018-02-05:
take a look to https://github.com/espressif/arduino-esp32/issues/387

hpwit, 2018-02-08:
@vindolin i have done it and it works like a charm thank you

baggior, 2018-02-08:
This tool depends on MKspiffs and it has to be recompiled with the new spiff configuration parameter introduced in the esp idf sdk, starting from 19/12/2017

So the issue is on mkspiffs now

usmanshahid001, 2018-04-02:
hi please help i had solve this issue earlier but i changed my laptop and finding the same problem...even replacing the same spiffs.a file in sdk and using the same sketch uploader...........but nothing is working...is there any difference for 32 bit window and 64 bit........

rfestag, 2018-04-02:
The correct solution is to update your esp32-arduino library to the latest version from git, and to use the get.py script (details in the esp32-arduino install instructions) to pull in the latests mkspiffs script (I think it is 0.2.2?). That is used by this plugin to do the transfer. Do not use the old spiffs.a.

usmanshahid001, 2018-04-03:
it worked with old spiff.a and mkspiffs....thanks rfestag for feedback

LongmontRobotics, 2018-04-14:
I deleted and reloaded the entire ESP32 code.
Does not work, Is there a valid windows install that someone can post.