me-no-dev/arduino-esp32fs-plugin #8: SPIFFS mount failed with error: -10025

robertgregor, 2018-03-15:
Hello, I have tried to put to my ESP32 SPIFFS the jquery pages, however, during the upload, I have this error:

[SPIFFS] data   : C:\RH\Sources\RH\RemoteHomeArduinoSketches\WiFi_ESP32\PirSensor\data
[SPIFFS] start  : 2691072
[SPIFFS] size   : 1468
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/favicon.ico

/jquery/dataTables.jqueryui.min.js

SPIFFS mount failed with error: -10025

SPIFFS_write error(-10010): unknown

error adding file!

Error for adding content from jquery!

[SPIFFS] upload : C:\Users\gregorro\AppData\Local\Temp\arduino_build_971420/PirSensor.spiffs.bin
[SPIFFS] address: 2691072
[SPIFFS] port   : COM24
[SPIFFS] speed  : 921600
[SPIFFS] mode   : dio
[SPIFFS] freq   : 80m

esptool.py v2.1

Connecting........_

Chip is ESP32D0WDQ6 (revision 1)

Uploading stub...

Running stub...

Stub running...

Changing baud rate to 921600

Changed.

Configuring flash size...

Auto-detected Flash size: 4MB

Compressed 1503232 bytes to 4190...

Writing at 0x00291000... (100 %)
Wrote 1503232 bytes (4190 compressed) at 0x00291000 in 0.1 seconds (effective 150323.3 kbit/s)...

Hash of data verified.

What can be wrong???

arskanov, 2018-05-22:
I can chime in with almost the same error..
```
[SPIFFS] data   : /home/aramanen/work/wifi-race-clock/data
[SPIFFS] start  : 2691072
[SPIFFS] size   : 1468
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/index.html
SPIFFS mount failed with error: -10025
[SPIFFS] upload : /tmp/arduino_build_847234/wifi-race-clock.spiffs.bin
[SPIFFS] address: 2691072
[SPIFFS] port   : /dev/ttyUSB0
[SPIFFS] speed  : 921600
[SPIFFS] mode   : dio
[SPIFFS] freq   : 80m

esptool.py v2.3.1
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1503232 bytes to 49354...

Writing at 0x00291000... (25 %)
Writing at 0x00295000... (50 %)
Writing at 0x00299000... (75 %)
Writing at 0x0029d000... (100 %)
Wrote 1503232 bytes (49354 compressed) at 0x00291000 in 8.4 seconds (effective 1425.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```
Followed instructions from README.

arskanov, 2018-05-24:
@robertgregor The issue is probably that your file is too long. 32 bytes max for the path length is a default. I can't find the link right now but it's talked about in the esp-idf repo.

Dan-deWaal, 2018-07-31:
I am having the same trouble as @arskanov original problem.

I don't see how adding a SPIFFS.begin() to the .ino (c code) would help the external tool?

It was working for me a few weeks ago, and as far as I can think, nothing has changed or been updated. I rebooted all devices (including PC). I opened old sketches that were working for me those weeks ago, but they have the same error: SPIFFS mount failed with error: -10025

I'm using Arduino 1.8.5
My esp32fs.jar file is dated 30 Sep 2017

	[SPIFFS] data   : Y:\ISOBUS\ESP32 arduino code\librarytester\data
	[SPIFFS] start  : 2691072
	[SPIFFS] size   : 1468
	[SPIFFS] page   : 256
	[SPIFFS] block  : 4096
	/Spread_Spray_200.iop
	/Spread_Spray_200.iop.h
	/Spread_Spray_bak.iop

	SPIFFS mount failed with error: -10025

	[SPIFFS] upload : C:\Users\ddewaal\AppData\Local\Temp\arduino_build_920707/librarytester.spiffs.bin
	[SPIFFS] address: 2691072
	[SPIFFS] port   : COM10
	[SPIFFS] speed  : 921600
	[SPIFFS] mode   : dio
	[SPIFFS] freq   : 80m

	esptool.py v2.3.1
	Connecting.....
	Chip is ESP32D0WDQ6 (revision 0)
	Features: WiFi, BT, Dual Core
	Uploading stub...
	Running stub...
	Stub running...
	Changing baud rate to 921600
	Changed.
	Configuring flash size...
	Auto-detected Flash size: 4MB
	Compressed 1503232 bytes to 9371...

	Writing at 0x00291000... (100 %)
	Wrote 1503232 bytes (9371 compressed) at 0x00291000 in 0.1 seconds (effective 82936.9 kbit/s)...
	Hash of data verified.

	Leaving...
	Hard resetting via RTS pin...

closed 2019-01-02 by robertgregor