me-no-dev/arduino-esp32fs-plugin #13: java.lang.NumberFormatException: For input string: ""

bootrino, 2019-02-04:
I get the following error trying to upload using this plugin:

Any clues valued.

```
Arduino: 1.8.8 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Default, 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

java.lang.NumberFormatException: For input string: ""
	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
	at java.lang.Integer.parseInt(Integer.java:592)
	at java.lang.Integer.parseInt(Integer.java:615)
	at com.esp32.mkspiffs.ESP32FS.parseInt(ESP32FS.java:156)
	at com.esp32.mkspiffs.ESP32FS.createAndUpload(ESP32FS.java:242)
	at com.esp32.mkspiffs.ESP32FS.run(ESP32FS.java:378)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
For input string: ""
```

bootrino, 2019-02-09:
I found the problem.

It happens when `./Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/tools/partitions/default.csv` contains this:

```
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M,
storage,  data, spiffs,  ,        0xF0000,
```

Looks like the partition parsing code in this utility can't handle that partition structure for some reason.

DarvosBR, 2019-04-02:
@bootrino , first of all, huge fan of your work! especially with the esp32 a2dp source project!
Now let me ask, did you find a solution for this tool with the modified partition? if no, there is any other similar tool available! Thank you

DarvosBR, 2019-04-02:
@bootrino . Well I just returned the lib its original content, use the tool and its working perfectly! Ty!

bootrino, 2019-04-03:
@DarvosBR thanks for the kind words.  I fixed it in the end by modifying the esp32 partition csv file - I suspect the problem is this utility cannot parse the line that says "1M" but I never bothered digging any deeper.

RJ-Fynydd, 2019-05-24:
Could it be happening when the offset is empty?

RJ-Fynydd, 2019-05-24:
I can confirm it does work when the offset is not blank. 
I used this command to convert the partition table with blank offset values to a bin:
`python gen_esp32part.py input_partitions.csv binary_partitions.bin`

And then verify the bin file just created with the command above:
`python gen_esp32part.py binary_partitions.bin`

And jsut like that it will print the computed offsets in console for you. You can then replace the csv file with the new values computed.

DarvosBR, 2019-06-25:
Guys I found the solution, at least in my case, by using mkspiffs tool provided by espressif, I made a new partition csv but in order to use the mkspiffs to upload the WAV file I made a batch script to run the mkspiffs with the memory locations I need.