me-no-dev/arduino-esp32fs-plugin #15: SPIFFS Error: esptool not found!

petrkoc, 2019-02-23:
Hello. The latest version of ESP32FS (file ESP32FS-1.0.zip dated 2019/01/15) does not work, showing error "SPIFFS Error: esptool not found!" when trying to upload data.

I am using Arduino IDE 1.8.8 and Arduino core installed through "Instructions for Windows" during this week.

pieterverhees, 2019-02-27:
Same here. 
I noticed the file has the extension .java and not .jar.
Not into java and jar but I'll do my research.

tnx me-no-dev and others for the great work you share!

pieterverhees, 2019-02-27:
Okay, tried to compile from .java to .jar but it seems to lack some dependencies. Any help will be appreciated.

pieterverhees, 2019-02-27:
Think is solved for me. Found it here: https://github.com/me-no-dev/arduino-esp32fs-plugin/releases/

pieterverhees, 2019-02-27:
More on this: the new version doen't work indeed! "SPIFFS Error: esptool not found!"
To solve for now I used the older version and that one still worked!

johnty, 2019-03-14:
it seems like it doesn't work if you installed the toolchain via github. if you installed arduino-esp32 via the board manager, then it finds esptool.py without issues. so i think it has to do with where it assumes esptool.py resides...

this is on OSX, using IDE 1.8.5.

nouser2013, 2019-03-16:
Am on windows 7 with Arduino IDE 1.8.9 and had the same problem. Tried installing esptool via system wide python ("pip install ...") to no avail.

Seems like a path issue. In the Java-File, in line 276 and following, the tool determines the path to the ESP-tool platform-dependently to be ```platform.getFolder()+"/tools", esptoolCmd``` with ```esptoolCmd``` presumably resolving to ```esptool.exe``` on Windows.

Looking into the folder, there's just an ```esptool.py``` there which can be executed, of course. But, as the Arduino IDE plugin expects an ```*.exe```, this doesn't work. 

Luckily, there is the binary under the same folder in the ```esptool``` subfolder, so just copy the file ```platform.getFolder()+"/tools/esptool/esptool.exe``` --> ```platform.getFolder()+"/tools/esptool.exe``` next to the ```esptool.py``` python script.

Works for me :)

SimonMerrett, 2019-04-29:
> More on this: the new version doen't work indeed! "SPIFFS Error: esptool not found!"
> To solve for now I used the older version and that one still worked!

I used version 0.1 when I encountered this issue and it seems to work. For info, Arduino 1.8.7 on Win10 and ESP32 installed in boards manager with https://dl.espressif.com/dl/package_esp32_index.json

zuqualla, 2019-06-03:
after a long search i found i had two "tools" folders (win 10):

1. C:\Users\[User Name]\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\tools
and 
2. C:\Users\[User Name]\AppData\Local\Arduino15\packages\esp32\tools

i had to copy the "esptool.exe" from a subfolder of the second, to the first mentioned folder :) 

or with a good search tool like "everything" (forget about win search):
- just search for "esptool.exe", copy this file, 
- and search for "esptool.py", go it's containing folder, paste "esptool.exe" here. 

wasted an hour on this :'( but hope to save someone else some time ;)

mbenitez01, 2019-06-19:
ES UNA BOSTA!!! NO ANDA NADA!! SOLO UN ITALIANO PUEE FABRICAR UNA BASURA TAN GRANDE!!! AVANZAN PARA ATRAS!! LA VERSION 1.6 ANDABA PERFECTA!! TODO TIENE ERRORES! EL FS.ZIP..EN LUGAR DE DESCOMPRIMIR UN JAR EJECUTABLE, DESCOMPRIME UN .JS (JAVASCRIPT) UN DESASTRE!!

sairfan1, 2019-06-22:
In my case i came to know something really important, addon must be inside the Sketchbook folder that is set in Preferences tab, ie. sketchbook\tools\ESP8266FS\tool\esp8266fs.jar also take care of version of IDE and Addon if both are compatible with each others.

Darkcast, 2019-07-19:
### **Solution** 
Windows 10
running esp32 devkit 1v 
running ardruino 1.8.9

Copy the file **esptool.exe** 
**from** C:\Users\%USERNAME%\AppData\Local\Arduino15\packages\esp32\tools
**to** C:\Users\%USERNAME%\Documents\Arduino\hardware\espressif\esp32\tools

outoftrun, 2019-09-19:
I had loaded some of the tools sets from various manufacturer of esp8266 boards Heltech, etc and seem to have copies to esptools everywhere, Ardunio seems to become very confused and the tools no longer worked.  Had to purge them all and reload slowly.

klkl140, 2020-01-18:
a solution for OSX 10.12
Arduino 1.8.10

rename the folder
/Users/----/Documents/Arduino/hardware/heltec/esp32/tools/esptool
to something like
/Users/----/Documents/Arduino/hardware/heltec/esp32/tools/esptool_away

and move the file esptool located in this folder to
/Users/----/Documents/Arduino/hardware/heltec/esp32/tools/

kevincshare, 2020-08-03:
Met the same issue for Arduino IDE1.8.13. The project image can be upload successful but failed with "ESP32 sketch the data upload". It shows error message "mkspiffs not found" when used 1.0 version. After copy mkspiffs.exe to the right tools directory, it shows error message "esptool not found". After copy esptool.exe similar, it go on and see the SPIFFS info but failed with "SPIFFS failed.".  Aha.........
My last solution is to use v0.1 in tags instead and it works.

praveen2k10, 2020-08-06:
https://github.com/esp8266/arduino-esp8266fs-plugin/files/2936951/ESP8266FS.zip
The above zip gets you past this error, however I an now getting :
[SPIFFS] python   : python.exe
[SPIFFS] uploader : C:\Users\Praveen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\tools\upload.py

Python was not found but can be installed from the Microsoft Store: https://go.microsoft.com/fwlink?linkID=2082640SPIFFS Upload failed!

praveen2k10, 2020-08-06:
Got the python issue resolved as well with 
https://github.com/esp8266/arduino-esp8266fs-plugin/releases/download/0.5.0/ESP8266FS-0.5.0.zip

mytechnotalent, 2020-08-16:
Thank you all!  I did the same as praveen2k10 and it works.

spyder0069, 2020-11-17:
On windows 10 with the files from github I had to copy esptool.exe from:
\SKETCHBOOK\hardware\espressif\esp32\tools\esptool
to:
\SKETCHBOOK\hardware\espressif\esp32\tools

Then it worked.

jpasqua, 2020-11-25:
I was also having the "esptool not found" problem with:
* arduino-esp32fs-plugin v1.0
* macOS Catalina (10.15.7)
* ESP32 installed from github - *not* using Board Manager as described [here](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/mac.md)
* Arduino IDE 1.8.13

Reverting back to v0.1 of the plugin cleared up the problem. I don't have an environment set up that allows me to build the plugin, otherwise I'd try to find out more precisely what is going on. Are there instructions anywhere on setting up such an environment?

agordeev, 2020-12-31:
Same here. Doesn't work on macOS Big Sur.

andres26263, 2021-01-13:
Also on Win 10 & Arduino IDE 1.8.13 (clean Install & Libraries updated & boards fresh reinstalled) have the same error...

lorol, 2021-01-13:
@andres26263 
For Windows, the plugin searches for esptool.exe in:
/tools
/tools/esptool_py
/tools/esptool
Then in values configured by installation
runtime.tools.esptool_py.path=
runtime.tools.esptool.path=

You can test my variant of the plugin, it gives a bit more info where the tool is found.
https://github.com/lorol/arduino-esp32fs-plugin

andres26263, 2021-01-15:
**@lorol**
Thank You for answer. I finally solved it, after 1 day of searching and trying, copying a newly redownloaded "esp8266fs.jar" in all "/tools" directory & subs.
This because of Steph answer in [https://rntlab.com/question/spiffs-error-esptool-not-found/]
Actually I'm working on a 8266 project, so I could'nt try Your variant, sorry...

tignioj, 2022-02-01:
> 

Simply move espTool.exe next to espTool.py work for me.

ybchen2005, 2022-02-28:
Just  move esptool.exe from  "C:\Program Files (x86)\Arduino\hardware\espressif\esp32\tools\esptool" 
                                          to "C:\Program Files (x86)\Arduino\hardware\espressif\esp32\tools"

somBabu1996, 2022-09-28:
I have the same issue. 
but I got the solution with this updated jar file.

[https://github.com/esp8266/arduino-esp8 ... -0.4.0.zip](https://github.com/esp8266/arduino-esp8266fs-plugin/releases/download/0.4.0/ESP8266FS-0.4.0.zip)

moononournation, 2023-01-23:
For arduino-esp32 2.0.6 in macOS, my resolve method is:
```
/usr/bin/pip3 install esptool
ln -s ~/Library/Python/3.9/bin/esptool.py ~/Library/Arduino15/packages/esp32/tools/esptool_py/4.2.1/esptool.py
```

maddisondesigns, 2024-01-01:
Adding my 2cents worth...
The only solution that worked for me was to install version 1.0.1 from @lorol's repo ([can found here](https://github.com/lorol/arduino-esp32fs-plugin/releases/tag/1.0%2C1)).

[Version 2.0.7](https://github.com/lorol/arduino-esp32fs-plugin/releases/tag/2.0.7) from the same repo wouldn't work.
And [this version](https://github.com/me-no-dev/arduino-esp32fs-plugin/releases/tag/1.1) didn't work either.

Apart from version 1.0.1 mentioned above, all the other versions I tried would either not work (due to the 'esptool not found' error) or corrupt the LittleFS partition even when they reported that the upload worked.

(And as I'm sure you're aware, this only works in Arduino IDE v1.x)