me-no-dev/arduino-esp32fs-plugin #19: osx installation of tool

pappgeza, 2019-06-16:
Hi,

I have osx and there is a problem if the sketchbook directory has changed:
in the description: _On the OS X create the tools directory in ~/Documents/Arduino/ and unpack the files there_

The directory name should be changed to "Setup->sketchbook location" in the description

echelon75, 2019-11-30:
Hi,

>The directory name should be changed to "Setup->sketchbook location" in the description
Please could you give us more details ?

Regards

Richard

stoune, 2020-12-29:
In my case it only worked when I put the plugin into /Applications/Arduino.app/Contents/Java/tools/ESP32FS

MShokry, 2021-01-08:
create a folder esptool_py and copy esptool inside it, 
It works fine with me
./Arduino/hardware/espressif/esp32/tools/esptool_py <= inside the esp32 folder