me-no-dev/arduino-esp32fs-plugin #20: Cannot OTA on 'fresh' ESP32-CAM

mozgy, 2019-07-16:
On a 'freshly unpacked' ESP32-CAM (aka AI-cam) SPIFFS OTA upload is not working, Arduino IDE says 'No response'.
Steps -
- compile ota sketch and prepare data dir
- ftdi upload sketch
- disconnect ftdi and reset board
at this moment further OTA sketch uploads work fine but SPIFFS OTA not at all - No response
- reconnect ftdi (gnd-IO0-reset)
- upload SPIFFS tru COMxx - works fine
- disconnect gnd and IO0, reset
at this moment all OTA uploads work fine (leaving or disconnecting ftdi does not matter)

repeatable on all AI-Cam boards I could find (over 10 of them)..

fah, 2019-07-26:
Where did you read that this will work? 
OTA service has to run first. (Except that you got the chip flashed already with something more then "blink".)
So an initial, serial upload is needed with a basic OTA script. Then you can start with OTA.

mozgy, 2019-08-23:
@fah - you haven't read the text carefuly - sketch OTA *is* working, SPIFFS OTA part is *not*, so yes, OTA service *IS* running !

szerwi, 2020-03-26:
Same problem here. Did you find out if it's possible to upload to SPIFFS via OTA?