me-no-dev/arduino-esp32fs-plugin #25: esp32s2

zyx0908, 2020-09-05:
[SPIFFS] data   : C:\Users\zyx\Desktop\sketch_aug29a\data
[SPIFFS] start  : 2686976
[SPIFFS] size   : 1472
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/index.html
[SPIFFS] upload : C:\Users\zyx\AppData\Local\Temp\arduino_build_540398/sketch_aug29a.spiffs.bin
[SPIFFS] address: 2686976
[SPIFFS] port   : COM3
[SPIFFS] speed  : 921600
[SPIFFS] mode   : dio
[SPIFFS] freq   : 80m

error: Invalid argument or value after --chip (argument #2)
SPIFFS Upload failed!

etherfi, 2020-12-07:
Try this fork:

https://github.com/etherfi/arduino-esp32fs-plugin-esp32s2

lorol, 2020-12-15:
or this one :) https://github.com/lorol/arduino-esp32fs-plugin

lorol, 2020-12-15:
Or wait a blessed one :) to come from upstream:
 https://github.com/me-no-dev/arduino-esp32fs-plugin/pull/28