me-no-dev/arduino-esp32fs-plugin #27: Can I use this plugin to create sketch+data firmware (.bin)

denismiiix, 2020-11-05:
How can I create a .bin file that I can pass to a third party user so that he can fully update the arduino esp32 without using any development environment - that is, no special programs and no hassle, the best way is to execute .bat?

rafaelbonventi, 2020-11-10:
hey Dnismiix where you able to get this code going? i am getting errors
20:12:22.673 -> configsip: 0, SPIWP:0xee
20:12:22.673 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:12:22.673 -> mode:DIO, clock div:1
20:12:22.673 -> load:0x3fff0018,len:4
20:12:22.673 -> load:0x3fff001c,len:1216
20:12:22.673 -> ho 0 tail 12 room 4
20:12:22.673 -> load:0x40078000,len:9720
20:12:22.673 -> ho 0 tail 12 room 4
20:12:22.673 -> load:0x40080400,len:6352
20:12:22.673 -> entry 0x400806b8
Start
 
black screen and it does nothing can you please help me ? thank you

lorol, 2020-12-06:
You may check this: https://github.com/krdarrah/ESP32_GUI_Programmer

lbernstone, 2020-12-15:
You can use esptool directly for this:
`esptool.exe --baud 921600 read_flash 0 0x400000 full.bin`
and then program the device with:
`esptool.exe --baud 921600 write_flash 0 full.bin`
Your issue is going to be handling the countless number of problems with Windows installations.