me-no-dev/arduino-esp32fs-plugin #29: pyserial error when trying to use tool on OSX Big Sur

mikevanis, 2021-01-08:
When trying to use the tool on Mac OSX Big Sur, esptool fails to execute. 

`
Traceback (most recent call last):
  File "esptool.py", line 57, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports.py", line 29, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_posix.py", line 31, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_osx.py", line 32, in <module>
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
Failed to execute script esptool
SPIFFS Upload failed!
`
I believe this is similar to the problem experienced with the arduino esp32 core (as mentioned [here](https://github.com/espressif/arduino-esp32/issues/4408)).

delcomp, 2021-02-03:
Did you find a solution for this issue?

mikevanis, 2021-02-03:
No actually, I've been getting by using [mkspiffs](https://github.com/igrr/mkspiffs) and esptool directly and forego this tool for now until it's fixed :)

vindolin, 2021-02-03:
Here's more info about it and a fix: https://github.com/pyserial/pyserial/issues/509#issuecomment-726966076

delcomp, 2021-02-03:
Ok, found my own solution. I have a virtual machine running 10.14 (a Parallels setup). Quickly installed Arduino IDE, the ESP32 library, this tool and a driver for the serial port. Then made a shared folder between the virtual machine and my Big Sur os. Now I can use the 10.14 to write the SPIFFS of my ESP32. 

Once setup it's fast to work with.

lorol, 2021-02-03:
You can check the last PR at my clone of this repository. I may be helpful for you.

delcomp, 2021-02-05:
Thanks @lorol, that did the trick.
Mind you, the '!Erase Flash!' option still gives an error. All the other options do work now.

Erasing all Flash started...
Port: /dev/cu.usbserial-1413420

_>Traceback (most recent call last):
_>  File "esptool.py", line 57, in <module>
_>  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
_>  File "serial/tools/list_ports.py", line 29, in <module>
_>  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
_>  File "serial/tools/list_ports_posix.py", line 31, in <module>
_>  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
_>  File "serial/tools/list_ports_osx.py", line 32, in <module>
_>ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
_>Failed to execute script esptool
!Erase Flash! Upload failed!

lorol, 2021-02-06:
@delcomp, thank you for the report. Probably @bergdahl would be able to help fixing it or give some hints.

bergdahl, 2021-02-06:
Basically the fix I made was to remove the use of the old "esptool" binary file and make it use "esptool.py" instead. I also changed the call to "python3 esptool.py" as Python 2 is obsolete.
The underlying problem is a change in Big Sur that forces and update to Pyserial.