me-no-dev/arduino-esp32fs-plugin #33: Plugin does not use custom partitions.csv

dl9rdz, 2021-09-26:
Since https://github.com/espressif/arduino-esp32/pull/3328, there can be a custom partitions.csv file in a sketch.

The Arduino ESP32 filesystem uploader plugin ignores such custom partitions.csv. Thus it does not use the correct size and location for the SPIFFS for Arduino sketches that use a custom partitions.csv file.

robertgregor, 2022-12-05:
I am also looking into this thing. @dl9rdz you have already the working jar, which is using it? Can you send it to us? Seems, me-no-dev didn't merged your code yet :)

dl9rdz, 2022-12-05:
It should be here:
https://github.com/dl9rdz/arduino-esp32fs-plugin/releases/tag/1.0mod

robertgregor, 2022-12-06:
Hi, thanks a lot. I have tried to do modification in the source code and tried to compile it with the javac, but I don't know, which jars from arduino, I have to include to the classpath. Can you add the javac compile command here? Or are you using some arduino tool to compile the java file? Thanks!!!

dl9rdz, 2022-12-06:
Just running "make.sh" works for me.

javac needs pde.jar, arduino-core.jar, and commons-codec-1.7.jar from arduino

twinsvision, 2023-09-20:
Hi, with this esp32fs plugin I get the following error which was not the case with the unpatched version:
"A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?"

twinsvision, 2023-09-20:
@dl9rdz Great job! Working like a charm now, many thanks :)