me-no-dev/arduino-esp32fs-plugin #41: Unspecific "SPIFFS Upload failed!" - Ubunut 22.04

GaudiLabs, 2022-10-14:
Got an unspecific  "SPIFFS Upload failed!" just before esptool.py upload with no further error message running Ubunut 22.04. Found the problem to be a python vs Python 3 command call (and binary path) issue. Solution for me was:

sudo apt update
sudo apt install python-is-python3

https://linuxpip.org/python-is-python3/
Maybe this can help others...

e2002, 2022-10-22:
@GaudiLabs Thanks it works

mattymactutor, 2023-01-31:
life saver!!!! thanks

WagnerDeQueiroz, 2023-04-02:
@GaudiLabs This Solution saved my life too! 
I cannot write my code with VSCode, but now with arduino I can upload, thanks!

phpbbireland, 2023-04-28:
Worked like a charm...

fexmak, 2023-09-10:
this really helped! thank you!

knoxsp, 2023-12-14:
Spent days chasing this problem. This solution worked! Thanks!

fjrg76-com, 2024-07-12:
Great! Why me-no-dev hasn't include this workaround into the installation instructions?