me-no-dev/arduino-esp32fs-plugin #44: When is support for Arduino 2.x.x likely to happen .. thanks

bmentink, 2023-02-06:
Great tool, looking forward to Arduino 2 support.

linotex, 2023-02-07:
I am waiting too  :D

thrmh, 2023-02-11:
I'm waiting

electrolabopen, 2023-03-17:
Im waiting :(

me-no-dev, 2023-03-20:
If you are talking about IDE 2.x support, then that depends currently largely on the Arduino IDE team to add support for external plugins. If you are waiting for the plugin to support the new chips, then that has already been done last week (check the latest version)

ilai23, 2023-08-22:
Is there any new information about the IDE 2x support for external plugins?

thrmh, 2023-08-22:
no more

---- Replied Message ----
| From | ***@***.***> |
| Date | 08/22/2023 23:36 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [me-no-dev/arduino-esp32fs-plugin] When is support for Arduino 2.x.x likely to happen .. thanks (Issue #44) |

Is there any new information about the IDE 2x support for external plugins?

blake5634, 2023-10-06:
+1 for this.

qwerty13, 2023-10-20:
no way to make this tool standalone?

Stemby, 2023-11-23:
> Is there any new information about the IDE 2x support for external plugins?

You can follow [this](https://github.com/arduino/arduino-ide/issues/58) issue.

AjayGautam1199, 2024-03-15:
i am also facing this issue

FireflyJar, 2024-03-19:
I believe we need to open an issue with Arduino. but meanwhile...
I just have both versions, 1.8.19 and the latest v2, easy to swap between them.

per1234, 2025-01-24:
There is now an equivalent extension for Arduino IDE 2.x:

https://github.com/earlephilhower/arduino-littlefs-upload

Unfortunately it is not technically feasible to add support to Arduino IDE 2.x for the Java-based Arduino IDE 1.x ["Tools" system](https://github.com/arduino/Arduino/blob/master/build/shared/tools/howto.txt). However, the new extensions system is quite flexible and powerful (it is the same system used by VS Code, and in fact you can use the existing VS Code extensions with Arduino IDE 2.x).