me-no-dev/arduino-esp32fs-plugin #54: ESP32 Sketch Data Upload menu item is missing

Chris-656, 2024-10-05:
Dar all,

ESP32 Sketch Data Upload menu item item is missing after installation process as described
- Version IDE 2.3.3
- Windows 11

regards Chris

emblink, 2024-10-19:
Hi @Chris-656 

It seems this option isn’t available in the IDE. I’ve encountered the same issue in Arduino IDE 2.3.2.
However, the ESP32 Sketch Data Upload option is present in my regular Arduino editor (version 1.8.13), so perhaps that could help you. Best regards.

[image]
[image]

zirkeltraining, 2024-12-26:
please make a note in the readme that this plugin only works in the 'legacy' version of the arduino software. 
It is not obvious, would be a great tinesaver.

kellnerp, 2025-03-19:
The README.md file just says "supported versions" leaving it to the reader to figure out what that is.

leotordo, 2025-04-11:
hello guys....I'm Stefano and I've same problem. IDE 2.3.4 and I don't find tool on "tool" menu.
Did someone solved this issue?
thanks
Stefano

sanchosk, 2025-06-27:
Unfortunately, same here - Arduino 2.3.6 official appimage file, not showing up :(